polyend/TrackerBetaTesting #295: [v1.2] Samples disappearing

onlyruin, 2020-10-25:
## Hi, sorry you had trouble!

### Steps to reproduce

I just updated to v1.2, and a bunch of my samples are no longing showing up in the sample loader. All of my samples are or have been converted to 44.1 kHz and 16 bit, and they are showing up on the sd card when I put it in my computer. When I open old projects that are using the samples from before I updated the firmware, they appear to still be in the projects, but I can't find them anywhere in the actual sample folders.

### Attachments and links

Here is a quick video showing the samples on the sd card, then not reading on the tracker: https://youtu.be/4wjrJn8t-W0

### Other important information

* I have restarted the tracker, and this seems to be a permanent change.

wjakobczyk, 2020-10-25:
Could you share with us these problematic samples? That would help us find the root cause and fix the issue.

onlyruin, 2020-10-25:
> Could you share with us these problematic samples? That would help us find the root cause and fix the issue.

There are 1000s of them. Should I just share a few?

onlyruin, 2020-10-25:
Here's a zip of one of the folders that is not showing up. 
[AcidAmen.zip](https://github.com/polyend/TrackerIssues/files/5436011/AcidAmen.zip)

onlyruin, 2020-10-25:
After trying a few things, I think I've found part of the problem. It seems that by automatically breaking the folders up into multiple folders with 100 samples in them it's limiting how many samples are displayed overall. If I arrange everything differently and split them up into subfolders like "Drums" "Synths" etc, more show up, but it's still not showing all of them. Some of the subfolders have 1000+ samples in them, so it breaks those into 10 separate subfolders, which seems to be stopping it from showing the rest of the subfolders that already exist. I don't know if this makes sense, but I think it's part of what is happening.

wjakobczyk, 2020-10-26:
Thanks that will help!

wjakobczyk, 2020-10-26:
Could you also share one of these large folders with 1000+ files?

wjakobczyk, 2020-10-26:
We have found a potential fix for the issue, could you please give the folllowing build a try and let us know if the problem is gone?
[Polyend_Tracker_Firmware_1.2.1.zip](https://github.com/polyend/TrackerIssues/files/5438600/Polyend_Tracker_Firmware_1.2.1.zip)

onlyruin, 2020-10-26:
> We have found a potential fix for the issue, could you please give the folllowing firmware build a try and let us know if the problem is gone?
> [Polyend_Tracker_Firmware_1.2.1.zip](https://github.com/polyend/TrackerIssues/files/5438600/Polyend_Tracker_Firmware_1.2.1.zip)

These seems to have fixed it! Thank you so much!!

wjakobczyk, 2020-10-26:
Good to hear that, thanks for the help.

closed 2020-10-26 by wjakobczyk