polyend/TrackerBetaTesting #296: 1.2.0 Record on a muted track (step vs live rec)

stevesomeone, 2020-10-25:
I have noticed that when your focused track is muted, the recording behavior is different depending on if you are in step or live record mode.
If you are in step mode, the information is still recorded correctly.
If you are in live record mode, the information will not be recorded on a muted track, but the next available unmuted track.

* How often does this bug happen? Every time 

* Does this bug happen in previous firmware versions? It has happened in previous firmware versions, but I can't be sure when it started, or if it has always been this way.

* Does this bug happen in every project? Yes

Jacknametrouble, 2020-11-02:
Not sure it's a bug, but can confirm this particular behaviour.

shuler323, 2020-11-12:
In my opinion, it's correct behavior. In 'live rec', muting track is the way to deselect tracks from recording. In 'step edit mode' there must be way to set, delete, edit note, etc (or Instrument / fx) even if its track is muted. For now I will close this issue, feel free to reopen and put new arguments :) cheers

closed 2020-11-12 by shuler323