polyend/TrackerBetaTesting #299: [1.2.0] The event disappeared from all patterns while editing a pattern.

modmodmodmod, 2020-10-26:
About 30 patterns of events were all emptied in an instant.

I tried to restart it and when I pressed and held the power switch, the bar progressed, but I could not shut it down.

When I unplugged the power cable, it shut down, but when I started it, I was forced to insert the SD card even though the SD card was stuck.

The events that disappeared after repeatedly inserting and removing the SD card and starting and restarting has returned, so it's okay for the time being.

I couldn't keep a record because I gave priority to recovery and backup, but I will report it because there was a bug of # 294 in the same v1.2.0 yesterday.

I'm afraid that the errors I encounter in this version can lead to more data loss than ever before.

muhfugga, 2020-10-26:
I had a similar thing happen. I was editing patterns in pattern view, going back and forth between 2 patterns and then they both disappeared. I then could not shut it down unless I unplug the usb cord.

ambv, 2021-01-05:
Is this still happening in the current firmware? There's been rather large changes which likely fixed this problem.

ambv, 2021-01-17:
Closing, please reopen if you end up seeing this again.

closed 2021-01-17 by ambv