polyend/TrackerBetaTesting #300: Midi Bank Selection

BTourist, 2020-10-26:
## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

### A new feature
At the moment it is possible to send a midi program change to one external synth. However it does not appear possible to send a midi bank selection change to external synths that support this, ie synths with more than 128 presets. Is it possible to add this feature?

Please tell us about your feature in as much detail as you can provide.
Vague ideas are less likely to be successful.  Concrete requests are
easier to estimate.

Just-Peter, 2020-10-26:
Isn't this same feature request I placed?

https://github.com/polyend/TrackerIssues/issues/177

BTourist, 2020-10-26:
No, I think you were asking for program change per track, which I would also like to have, and have supported your request. 
This request is for external bank selection, ie several of my synths have banks of presets (A - E for example), each consisting of 128 presets. At the moment I do not think it is possible to select/change these banks. 
I would like to see it implemented on a per track basis as per your request. Hope this makes sense.

Just-Peter, 2020-10-26:
It makes sense. Would be great to have this.

modulationsource, 2020-10-28:
I think Bank Change can be done with Controller CC Number 0.
I did not try it, but Number 0 is often named "bank change"

BTourist, 2020-10-29:
Thanks for this suggestion. CC Number 0 did not work(msb), but CC Number 32 (lsb) did. 
The only downside is losing a cc command. Also needs to implemented on a track/step basis as per Just-Peter suggestion to enable multiple external gear to be controlled

Tomsketom, 2020-12-08:
This is something we are waiting for @Jacknametrouble

closed 2021-12-02 by Jacknametrouble