polyend/TrackerBetaTesting #301: Record Performance Audio

sean-muyaba, 2020-10-26:
Performance mode is very fun and simple to use, allowing you to extend a very simple endlessly however I think this could be extended to all work as a more dynamic way of arranging/composing a song

# # Use case 1
If it was possible to record the audio of each track while you perform you can then just grab these put them in a DAW and edit them into a more coherent song and also cut parts that didn't work live

# # Use case 2
I'm not sure about the HW limitations with regards to recording audio but alternatively, your performance could be automatically caputed into the next available pattern and placed in song mode in an increamenting order, keeping in mind that some perfomance actions may not be recordable, however even capturing 50% of the perfomance brings you much closer to a song much faster than the current song mode I think

### example
Patten 1 has a length of 8 and has 2 tracks, track 1 has instrucment 1 and track 2 instrument 2 both instrucments play on every second step

if I start my recording with track with track 2 muted the tracker will start capturing instrument 1 in the next available pattern (in this case pattern 2) and as I apply effects, these get added as FX to the pattern and when track 2 is unmuted, instrument 2 playback gets captured as its played back along with any FX that can be applied

tfspsound, 2020-10-27:
Would it be possible to make another screen in performance mode that allowed you to group tracks together that you wished to apply fx to and what fx combinations applied to them and saved them as 48 fx presets on the keypad? I like performance mode too so it would be cool to save good results from mangling so it needn’t always be so experimental.

jaredarnold, 2020-11-05:
I wrote out my thoughts on the same basic concept before realizing you had addressed this already. Including my thoughts below, but basically: another vote for this idea.

Currently, you can arrange a song and export stems in addition to the mix. However, while playing live and using performance effects, you're limited to recording the full stereo mix of your live tweaks.

Is it possible to implement a mode that would allow the recording/exporting stems of a live performance? I imagine this would be either

A way of live recording all the stems simultaneously in real time to the SD card. (Is the Tracker even technically capable of that?)
A way of recording in performance effects into the pattern interface directly. (Perhaps this would be in a 9th Master Track, or in an 8th track that, if in use, precludes normal instrument programming). Naturally there would be a limit to the number of performance effects that could be entered on a given step.
I understand that this idea might be a pipe dream, and that at best this is a longterm request. But FYI I'm finding myself very productive with the Tracker as a place to write songs, and then I polish the stems in a DAW. I would love to be able to work performance effects into that workflow.

modmodmodmod, 2021-04-29:
I'm amazed at the horizontal tracker sequencer updated in 1.4.0b1 and once again pay tribute to polyend's innovation. And I believe the great polyend will evolve Tracker so that we can record performance in perform mode on stems in the not too distant future. Like many other PT users :-)

closed 2021-12-02 by Jacknametrouble