polyend/TrackerBetaTesting #304: [1.2.1] micromove and roll combination issue.

tfspsound, 2020-10-27:
Sup trackers?

Firstly thanks polyend really enjoying the tracker fantastic piece of kit and glad I got one. 

### Steps to reproduce

Record a beat using live recording where a kick drum hits just before a snare on the same track so that the snare chokes the kick. Add a fast roll to the kick just before the snare and and the snare disappears and resets the roll to r4 or something similar. 

I re-recorded the same beat on a different track this time step recording and adding micro move values on each step and found the snare will choke the kick if it doesn’t have a micromove value attached to it. 

Top replicate make an 8 step pattern put kicks on step 1,4 and a snare on step 5. Add micromove to steps 4 (in my case m65) and 5 (m12)and add roll on step 4 (R8)

### Attachments and links

Non atm

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

Any micromove above 4 on the snare seems to do it. The snare chokes the kick with a roll value of 4 but anything else glitches 

* Does this bug happen in previous firmware versions?

1.1.23 so I updated to 1.2.1 and it still does it. 

* Does this bug happen in every project?

Not tried. 

Hope this helps. 

Cheers

Jacknametrouble, 2020-11-02:
I've recreated the exact mentioned scenario and in my understanding, it's not a bug, just a regular behaviour of the used step FXs.

Jacknametrouble, 2020-11-03:
Video confirming Roll + Micromove issue

https://photos.app.goo.gl/Zun8Zr4qRxgvvUGA9

closed 2021-01-05 by shuler323

tfspsound, 2021-01-05:
Nice 👍