polyend/TrackerBetaTesting #309: Advanced Swing Settings

jaredarnold, 2020-10-27:
(Note: I realize that users have already requested swing functionality, and that a version of swing is already possible by using Tempo FX.)

If/when swing is implemented, I'm very interested in seeing it done so in the most flexible way possible. I think the [implementation in Renoise](https://tutorials.renoise.com/wiki/Groove_Settings) is a great starting place. It lets you control not just the delay of each second line, but a repeating pattern of time offset for lines 2, 4, 6, and 8, which allows more complex grooves.

I think the next place to go from there would be to make this kind of swing system bipolar, and allow for reverse swing (where the 2nd note of each pair is longer than the first).

An even more advanced implementation of swing (probably a pipe dream but I figured I'd put it out there) is found in the Pioneer Toraiz Squid's "[Speed Modulation](https://www.youtube.com/watch?v=csspiOfCy5Q)" functionality. I would love to see something like this implemented, where there's effectively an LFO applied to tempo.

smitherines, 2020-11-18:
Swing is a gaping hole in my tracker life

jaap3, 2020-12-27:
This is related to #258