polyend/TrackerBetaTesting #314: [1.2.0] Inst Params 1/3 inaccessible

joelsimeus, 2020-10-29:
When pressing instrument params sometimes the first window with tune, volume etc becomes inaccessible and you can only view envelope/lfo and effect send windows. I think this happened to me after using the sends window (3/3) on an instrument.

It did not happen on previous firmware versions (to me).

It has happened twice since I updated. Not initially but after a short while (after flicking back and forth multiple times)

I have only tested it in one project.

modulationsource, 2020-10-29:
The send window is not 3/3, it opens pressing the effect button (rightmost in instruments page 1/2).
From the sends window, you can return with the back button (rightmost on the sends page).
I state this because it took me a while to get used to this, I found myself several times hitting the Instrument Parameter Button missing to get back to Instruments page 1. But it has to be done with the "back" button. Maybe this was the case with you too.

joelsimeus, 2020-10-30:
Ok I will try that! Thank you!
Skickat från min iPhone

joelsimeus, 2020-10-30:
Actually, the problem is not going out from the sends window. Its just that after you return to inst params you can only see 2/2 not 1/2.

Skickat från min iPhone

30 okt. 2020 kl. 05:05 skrev Joel Simeus <joelsimeus@hotmail.com>:

 Ok I will try that! Thank you!
Skickat från min iPhone

joelsimeus, 2020-10-31:
Oh wait! Yeah thats correct. You have to press back first before you can see both inst params first. Thanks.

Skickat från min iPhone

30 okt. 2020 kl. 05:05 skrev Joel Simeus <joelsimeus@hotmail.com>:

 Ok I will try that! Thank you!
Skickat från min iPhone

closed 2020-11-01 by Jacknametrouble