polyend/TrackerBetaTesting #319: [1.2.1] Freeze in Perform mode

MachineAgeVoodoo, 2020-10-31:
Hi. While in Perform mode, having all tracks active/selected (but also if only certain tracks), the tracker freezes into a constant beeping sound if I do the following: Activate Pattern Play Mode on for example "Backwards", and then play with various Step Repeats. Playing with these two effects alot almost always makes the system hang on a sound and become unresponsive. Before the real crash takes place, you can tell also that something isn't right about the sound that results from combining these two effects.

If it makes a difference, I set up my effects so that the first two columns were step repeats, and then the pattern play after that (also two columns of that one) And the specific values that made the crash the last time was "Backwards" combined with going from "3" step repeat to faster ones and back.

This happens every time. Also happened on 1.2.0

Jacknametrouble, 2020-11-02:
Been trying to replicate hard, with exact same settings as yours, no success.

tfspsound, 2020-12-07:
[1.2.1] I had a crash that sounds like this last night. Again using performance mode with step repeats in the first three columns. I was using individual hits across all tracks and using the step repeats quickly swapping between them. I didn’t have backwards on but I think I noticed the light on the pad corresponding to play mode Back lit. The tracker seems to hang on a repeat completely unresponsive. I had to pull the plug on it.

kxx303, 2021-02-06:
Have this bug on my unit also - in OS 1.3. Freezes while switching sample playback direction on several tracks activated in perfom mode.

tfspsound, 2021-02-06:
[1.3.0] Tracker seems to crash still when you hit continue in Song mode whilst using Pattern Playback Mode see to reverse. At the end of the current pattern (playing in reverse) the Tracker jumps to the next pattern but doesn’t play anything. Unit is still operable but no sound.

closed 2021-12-02 by Jacknametrouble