polyend/TrackerBetaTesting #320: MicroPitch is interesting, but can we add a regular Pitch shift?

MachineAgeVoodoo, 2020-10-31:
Hi. The title says it all, an FX that simply pitches in increments of semitones seems to be missing. Apologies if I am missing something obvious. A sample as it plays through the pattern, could be pitched up or down per step. Then not only by retriggering it on a new key. This could then also be used to achieve "pitch bend" type of effects.

Thank you for reading

ROM72, 2020-11-01:
Yes, a pitch envelope would be great and not just a finetune envelope.

dmt-hub, 2020-11-02:
Yes, please!

tfspsound, 2020-11-19:
Totally onboard with this, it’d save copying samples to repitch them too.

nthe, 2021-03-02:
I would that as well, but it looks like this one was already requested and rejected. 😞

Duplicate of #344

MachineAgeVoodoo, 2021-03-02:
its implemented already in current firmware

nthe, 2021-03-03:
I was reffering to pitch envelope feature in automation panel, or am I missing something? 🤔

I see that the ticket is just about pitch shift and not about automation.... sorry for confusion. 😐

Jacknametrouble, 2021-03-03:
This isn't possible due to the used hardware resources.

closed 2021-03-03 by Jacknametrouble