polyend/TrackerBetaTesting #323: Instrument management

keith-macleod, 2020-11-01:
Hi, just playing around with the instrument randomization effect and found myself wishing for 2 things:

- instrument reordering
- instrument duplication

I think they also offer a lot of utility beyond this isolated use-case, for instance if offered the option to update or not update the pattern instrument numbering, reordering provides a method to audition/globally change a sound across a project. Duplication would be great for making a copy of a sliced beat or other crafted sound with different filtering or modulation options.

As always, thank you guys for making the tracker and improving it. It's so much fun it should be illegal.

tfspsound, 2020-11-19:
How about if when in the Pattern screen you hold Instrument and scroll the sample list that appears on the right. When you are on the sample you wish to move you hold Shift and now as you move the jog wheel it moves that sample up or down the list while you hold Shift. When you let go the sample is dropped in the current slot. 
This method would shift groups of samples up or down as groups. To swap samples I would suggest while having the Instrument column selected you press Shift + (1 - 48) the sample you wish to move, then whilst still holding Shift you press (1 - 48) the destination or sample you wish to swap it with. 
Maybe to copy you could hold Rec instead of Shift?

keith-macleod, 2020-12-12:
After over a week, I finally noticed that you snuck instrument duplication into 1.3.0b2. That's totally awesome, thank you.

NoraCodes, 2021-04-03:
I came here to request this! Instrument duplication is extremely useful but I think being able to rearrange instruments - specifically, to swap them without duplicating into a third slot - would be great for larger projects.

Jacknametrouble, 2021-12-18:
The ability to save/load instruments as a sample with all its properties has been added in the newest 1.5 b1 beta.

closed 2021-12-18 by Jacknametrouble