polyend/TrackerBetaTesting #326: sample playback aliasing

phaelam, 2020-11-03:
noticing last night that some samples, that sound good on their own, once loaded and played back from the tracker have some pretty noticeable aliasing. these are sample I've added to my SD card. they are in the correct format for the the Tracker. no FX, FAD, or ENV. 

so I started a new project and loaded the same sample on its own. and it was fine. not sure what causes the samples to degrade. I was wondering if perhaps loading a sample while the Tracker is playing, perhaps, corrupts the sample? it happens regularly. but I wasn't noting what I was doing when the samples go bad. as I didn't expect it. so I will bang on that today to see if its random or if there are steps that cause this to happen.

here is an audio file with the issue.
0:00 - 0:06 = original sample
0:07 - 0:13 = Tracker output
[Tracker_sample_issue.mp3.zip](https://github.com/polyend/TrackerIssues/files/5482923/Tracker_sample_issue.mp3.zip)

piotr3500, 2020-11-03:
Did you try turning on the Anti-aliasing option in Config?

piotr3500, 2020-11-03:
@phaelam please also check yout Bit Depth settings in Master section.
It sounds like you have it set <12.

phaelam, 2020-11-03:
anti aliasing is on....and I really don't like bit reduction, I leave that at 16

vaeinoe, 2020-11-04:
Could you attach one of the exact original sample files you are loading that causes the issue, so one could try reproducing the issue with same sample?

closed 2021-12-02 by Jacknametrouble