polyend/TrackerBetaTesting #327: [1.2.1] Rendering audio selection with (4-bit) master bit reduction never seems to finish

vaeinoe, 2020-11-04:
**Steps to reproduce:** fill a pattern with sample/note data to taste, set master bit depth to 4 bit, select complete tracks from pattern, render to sample.
**What should happen:** Tracker plays the pattern, thinks a bit, saves it to a file.
**What happens:** Tracker plays the pattern, then seems to get stuck in an endless loop (well, at least 2-3 minutes until I ran out of patience and figured out it'll probably never finish)

Initially I thought this was a problem with a particularly long reverb tail being rendered, even the audio was silent, but the result was exactly the same when all reverb sends were silenced completely. Changing the bit depth to 16 bit made the render work again.

I'm not sure if there's some other variable involved, or if the issue starts at some certain amount of bit reduction, and I didn't have the time to try to reproduce this with eg. any other bit reduction settings. However, at least with the 2-track project I stumbled upon this issue with, it seems that rendering with master bit reduction at full 16 bits works, 4 bits doesn't. Maybe the latter messes up noise floor / silence detection?

jaap3, 2020-12-27:
It might be worth retrying this with firmware version 1.3.0, I believe the audio rendering code has been rewritten so maybe this was fixed in the process.

vaeinoe, 2020-12-28:
Will have to try it out, thanks! As far as I understand, the master bit depth reduction was removed in 1.3.0, and it's now a per-track effect, so maybe this one is not valid anymore anyway...

jaap3, 2020-12-28:
Ah, I didn't realize that master bit depth reduction was gone. If you want it back you could add a +1 (thumbs up reaction) to #529 to support that request.

shuler323, 2021-01-05:
Unfortunately, all problems with export BEFORE 1.3.0 (1.3.0b3 in betas) are out of date because of new export engine. Other problems with export please write in existing issues.

closed 2021-01-05 by shuler323