polyend/TrackerBetaTesting #331: Per-Effect Track Enabling for Performance Effects

jaredarnold, 2020-11-05:
Currently on the Performance Effects page, you select which tracks in your pattern should be effected, and than anything you do applies to those tracks. 

I propose an alternate mode where you can select different combinations of tracks to be processed by each effect. In the Red (record-enabled) mode of the Performance Effects page, the track selections you would make would only apply to the currently selected effect.

While this would be a little more complicated (or at least, require a little more setup initially), it would allow more flexible processing of different groups of tracks. You could, for instance, use beat repeat on a drum track while you pitch the bass up and down.

This could work as a mode that you turn on on the Config page.

tfspsound, 2020-11-05:
I do like this idea but I’d propose something a bit simpler where pressing the performance button brought up a second screen that worked similar to the polyend preset. By selecting one of the forty eight keys and going back to the first screen you select which effects are to be applied to the current state. You then go back into the second screen select another key and repeat the process. Each time the performance fx states are saved to keys on the second page. This would allow you to get instant changes without having to rely to much on memory. 

I do like the idea of being able to group effects but I can imagine that being a nightmare to produce and as yet I can’t visualise a way to effectively do this without things getting clunky. Still you never know what polyend might pull out their hat.

jaredarnold, 2020-11-05:
Yeah, what you're describing would work for me too -- anything that gives more flexibility to where effects are applied would be an improvement.

closed 2021-12-02 by Jacknametrouble