polyend/TrackerBetaTesting #332: Filter keytrack

Plieuw, 2020-11-05:
It would be great if the filter cutoff has key tracking. 

Allowing the filter to open/close depending on note value. 

A positive/negative value would be great. (like how the panning is implemented)

vaeinoe, 2020-11-05:
Co-signed. I was about to add a ticket for the same. It's already mentioned here but IMO deserves to be its own enhancement idea: https://github.com/polyend/TrackerIssues/issues/167

Of course one can "fake" this with track FX, but takes up one slot, and is a lot more tedious to do manually, than having a keytrack parameter in filter params.

jaredarnold, 2020-11-05:
Yep, this would be very helpful.  Given that there are only two FX slots per track, the more modulation/automation that can be built into the instruments themselves, the better.

ArtFluids, 2020-11-06:
I think filter tracking is sort of functionally the same as just rendering the sample with the filter on. Like if you pick the filter settings you like and render that to a new sample then you **have** a tracking filter. At least one that tracks pitch 1:1.

jaredarnold, 2020-11-06:
> I think filter tracking is sort of functionally the same as just rendering the sample with the filter on. Like if you pick the filter settings you like and render that to a new sample then you **have** a tracking filter. At least one that tracks pitch 1:1.

Sure, but then you're locked in to that specific filter setting. It's nice to be able to tweak cutoff/resonance and hear what different settings of those sound like as they keytrack. And nice to be able to control those same settings via performance controls. I understand what you're saying, just would rather have a more flexible implementation that doesn't require rendering a sample every time.

vaeinoe, 2020-11-06:
Aside that, the rendering trick doesn't really work at all when you have filter envelopes, LFOs or other dynamic stuff going on aside just the key tracking (eg. you want the base cutoff to track pitch, and have an envelope / LFO modulating the result on top of that). If you render that, you have a LFO or envelope whose speed tracks the pitch as well whether you like it or not. (And resets at every note, but I suppose that's what happens with envs anyway)

And, of course as was mentioned, you can't set "amount" of key tracking with a rendered sample to a negative value or more / less than 1:1, in case you want to use the tracking for something else than "just" keeping the timbre similar regardless of playback pitch.

x2db, 2021-04-13:
Adding my weight to this one, besides filter there should be key track on amp so that a sample that sounds quieter in lower frequency would get a boost, so no need for Volume FX slot

tfspsound, 2021-04-30:
Still interested in this for the times I don’t want artefacts in my sounds in the low end withou sacrificing the high end.

closed 2021-12-02 by Jacknametrouble

colonel-blimp, 2021-12-15:
I was just about to request this same feature, when I noticed this was closed 13 days ago.

@Jacknametrouble Do you remember why this issue was closed?  It could help new folks like me determine if they should submit similar requests in the future.