polyend/TrackerBetaTesting #335: Slice Auto-Select by Pad

jaredarnold, 2020-11-05:
Love the Slice and Beat Slice sample playback modes. However, I think that the process of tweaking slice start times could be made a little smoother. Specifically, I find that having to continually switch back and forth between pressing the "Slice" and "Adjust" screen keys to be clunky.

What about having the pads automatically select their corresponding slice? That way, you could stay in Adjust mode and tweak start times more quickly.

As far as I can see, this doesn't break any existing functionality.

snesmeister, 2020-11-05:
+1 for this. Would make the process so much quicker. The only thing that would be potentially problematic is if you were playing a sample to hear where it ends, not where it starts.

However if you removed the 'Slice' button, and instead had a 'Start' and 'End' buttons and forced users to use the pads to select slices (which I think everyone would be happy with anyway), then you'd be onto a winner.

There's lots of functionality left on Shift too. Perhaps Shift + Pad could jump to the sample slice? That would be an easy fix.

vaeinoe, 2020-11-06:
A _huge_ plus / heart / me-too for this one. Selecting already mapped slices with pads would make adjusting the slices a lot faster and more intuitive. Especially when you automap something and end up with a huge amount of slices, it would just be easy to both select and audition a slice by pressing a pad, and then either adjusting start point or deleting what one hears.

Shift-pad is difficult in the sense that you can't do that with just one hand - it's not a huge issue if your right hand is already on the jog wheel for adjusting slice starts and you can press shift there, but still a tiny bit more cumbersome.

paalweb, 2020-11-06:
I think this already has been suggested here:
https://github.com/polyend/TrackerIssues/issues/217

vaeinoe, 2020-11-06:
> I think this already has been suggested here:
> #217

Yep, this one seems to be a duplicate of that.

jaredarnold, 2020-11-06:
> I think this already has been suggested here:
> #217

Thanks, I missed that one.. Closing this out.

closed 2020-11-06 by jaredarnold