polyend/TrackerBetaTesting #341: [1.2.1] Export not rendering song properly (gaps and skipped patterns

danapeks, 2020-11-09:
Hi there, 

As the title says really - tried to export a full song (not stems) and I tried twice, both times the audio skipped and it jumped about on patterns. Screenshot here so you can see the gap of one export:

`<a href="https://imgbb.com/">[image]`

Any help appreciated.

Thanks

jaap3, 2020-12-27:
Song export has been reworked for firmware version 1.3.0. If you haven't upgraded already I'd suggest doing so and trying this again. Hopefully your issue has been resolved. If this issue is still present, please report back!

shuler323, 2021-01-04:
@jaap3 is right again :)  other problems with exporting please write in existing open issues

closed 2021-01-04 by shuler323