polyend/TrackerBetaTesting #345: 2.0 Unable to export Stems/Song/Pattern

ttiill, 2020-11-10:
[image]

It didnt happen before it suddenly appeared. But now it seems to be the case for all Projects. In the previous Firmware i did not have this problem but it did not appear after the update to 2.0. 

Thank you for the support so far 👍

jaap3, 2020-12-27:
Assuming you've experienced this issue with firmware 1.2.0 (since there is no version 2.0). Please consider upgrading to firmware version 1.3.0. From what I've read song exporting has been rewritten, so the issue you had might be fixed. Please note that exporting songs can take a very long time, so please be patient when doing so.

shuler323, 2021-01-04:
@jaap3 is right, other problems with exporting please write in existing open issues

closed 2021-01-04 by shuler323