polyend/TrackerBetaTesting #350: 1.2.2 Sample recorder/playback/editor pitch problem

muhfugga, 2020-11-12:
Hello, I recently tried to update my firmware to 1.2.2 and the first firmware failed. I downloaded another one and it worked. But now when I sample in the sample recorder and also edit in sample edit window, the all white buttons used playback the sample play in low distorted pitches. I remember during one of the firmware updates the tracker got stuck in the update screen and i pressed the little reset button on the back of the unit. I don't know if I did something wrong when updating the firmwares that might have caused this but now it doesn't matter which firmware i use I have this issue with the low distorted pitching on all my recorded samples in those particular windows.

muhfugga, 2020-11-12:
Actually the problem effects anytime i try to chromatically pitch the sample in any window.

muhfugga, 2020-11-17:
I still can't pitch my samples after that 1.2.2 update. Like my machine is messed up after.

ambv, 2021-01-19:
Does it still happen with the 1.3 firmware? Did you try resetting the configuration of the device (Config/Firmware/Reset config)?

closed 2022-02-06 by MortalJS