polyend/TrackerBetaTesting #351: Braid oscillator engine

Roganis, 2020-11-12:
I've seen an implementation of the Braids (Mutable Instruments) oscillator in the M8, which made me understand it's quite an efficient engine that can run on very low power devices, maybe we could have some of that goodie in the Tracker? We already got a nice wavetable engine, but some mono VA stuff would be bonkers ahah
Also the code is open source so it doesn't seem that hard to implement? Of course I'm not knowledgeable enough to understand how much of a pipedream this is, but having those awesome oscillators would make the Tracker a truly astounding machine
The source code for it : https://github.com/pichenettes/eurorack/tree/master/braids

ArtFluids, 2020-11-13:
I'm sure it's way outside the intended scope for the Tracker but it would just be KILLER to have this
The Tracker would kick the butt of every Elektron machine at that point

djr005, 2020-11-20:
Careful, we're not allowed to criticise Elektron ;)
Pisses's me off how they cheaped out by not putting a song mode in the digitakt.

XANi, 2021-01-06:
@Roganis might be efficient for what it does but the original braids itself got pretty speedy CPU (72MHz IIRC) and M8 runs on 600MHz Teensy

I'd love if we just had few synth voices builtin but in the end I'd rather have kickass tracker/hardware DAW and drive synths by MIDI rather than split dev effort between making tracker part better and adding synths to it

mxactvtd, 2021-01-28:
Even just a simple Sine or preferable also Saw, Square, Triangle with frequency accurate Hz tuning would be amazing, does not be a fancy Braids, Serum or Surge, just a built in Tone-generator, even if as offline render, would do :)

XANi, 2021-01-29:
@mxactvtd well there is wavetable engine if you just want some sines :D

Jacknametrouble, 2021-12-18:
Thank you for sharing your idea. This won't be possible due to the limited resources of the instrument.

closed 2021-12-18 by Jacknametrouble