polyend/TrackerBetaTesting #352: Tables

Roganis, 2020-11-12:
## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

### A new feature

Please tell us about your feature in as much detail as you can provide.
Vague ideas are less likely to be successful.  Concrete requests are
easier to estimate.

sunegg1024, 2021-01-06:
Just want to throw some more weight behind this and include that, should you implement it, it would be great to have tables both assignable to an instrument but then also called from an effect in Pattern like with a Txx (as you can do in LSDJ). The table effect would override the instrument table (as with LSDJ).

brandonlogic, 2021-01-09:
I agree this is a great idea and an important feature of a tracker.
To expand on this idea, id would be good to also the option whether the tables steps retrigger the envelope or not.  for example, if the retrigger was off the arpeggio would keep running as the envelope closes.

werd4311, 2022-01-15:
Still waiting and hoping. This baby will never be complete without tables. Please?