polyend/TrackerBetaTesting #353: Expand and Shrink Pattern

djkurzman, 2020-11-12:
This was mentioned in a comment to another request request. I think it's a good idea so I wanted to set it for it's own post.

A handy feature for the pattern editor would be to expand and shrink a selection by x2 or 1/2 steps.

For example:

1. **note**
2. ...
3. **note**
4. ...
5. ...
6. ...
7. **note**
8. ...

can shrink to 

1. **note**
2. **note**
3. ...
4. **note**

And likewise, expanding the selection would double the pattern by adding spaces between each step

ArtFluids, 2020-11-12:
Expand/shrink is already a feature

djkurzman, 2020-11-12:
Whaaaa??!!! Probably one of the many new features that totally snuck by me. Thanks for pointing that out!

closed 2020-11-12 by djkurzman