polyend/TrackerBetaTesting #355: New Idea for future releases in Song Mode

lugokeT, 2020-11-13:
Hello,

a great future function would be to select individuelly for 8 Tracks the Pattern Number in Song mode.
For Example in Songblock Number 1 when pressing play Track1 pattern2 is playing in Track2 pattern4 is playing so we would be able to play different Patterns for each individuell Tracks and combining just would get so easy and so i think the workflow could changed positvely.

And another plus future for this idea would be if we are able to set different tempos toghether.
For example Track1 pattern 2 is playing with lets say tempo 128 bpm and Track2 pattern 4 is playing with double the speed 256 bpm and booth of them played simultaneously on Song Block with Master Tempo 128.

This would be really Game Changer for Polyend Tracker

Regards

Lugoket

mFoxRU, 2020-11-22:
Yeah, I kinda also prefer lsdj-style song mode, where in song slots you can select each track pattern individually. No need to copy same kick/snare/bass track to almost every new pattern. No need to create new pattern only to make a minor change in one track. This approach shines in lsdj's live mode, where you can change individual track's patterns as the song plays, making jams and live performances much more dynamic and fun.

Not sure how this will work with current design, but it would definitely enhance workflow.

zshall, 2020-12-30:
Plus 1 for this; the ability to mix and match tracks from different patterns in song mode would make it a lot easier when writing drums and melodies separately. Suppose you make a drum track and copy it across many patterns and then you decide to change the drums, you’d need to go into each track and change them.

Something to consider is how the lengths of different patterns would work. Suppose one pattern has 16 steps and another has 32... would you want the 16 step pattern to repeat to fill the whole 32 steps or would you want the 32 step pattern to cut off after 16? The answer could be to have a length attribute to each slot in song mode. Everything would loop until the length of steps in the song mode slot was done. This could also cut down on the number of slots you’d need since if you wanted a 32 step pattern to repeat 4 times you could set the length of the slot to 128.

dimatura, 2021-04-07:
Also +1. I think just allowing each track in a slot to come from a different pattern, rather than something more elaborate like LSDJ would make sense given how the PT works now. I used this a lot with Buzz tracker (though that one's song mode is actually more elaborate).
As for different pattern lengths, I would be OK with either looping the shorter patterns or just playing it once . I'd work around it either way, eg, just making all patterns the same length and copy/pasting as needed. Of course, having the option of looping or not would be better, but I could live without it.