polyend/TrackerBetaTesting #359: [v1.2.2] n – Random note problem

PaulAntioche, 2020-11-13:
hello.
in the tracker notice it says this:
**in The FX step effects list:**
"n – Random note, sets the range of randomization for Note with each sequencer pass. From +/- 0 to +/- 100. Working accordingly to the global musical scale selected for the pads in Config."
but **in fact it does not work**. When I change "pads scale" in "general config" nothing changes. the notes played at random do not follow the type of game selected.

if the problem is solved **I suggest that the choice of the game "pads scale" is regulated in each patern** and not in "config".
that would be very interesting.

### Other important information

* How often does this bug happen? every time, 

* Does this bug happen in previous firmware versions? always

* Does this bug happen in every project? yes

77377, 2022-05-19:
Is this gonna be fixed or what? It renders one of the most powerful sequencer features pretty much unusable. /:

PaulAntioche, 2022-05-21:
Yes, this has been fixed with recent versions.

closed 2023-04-13 by MiroPoly