polyend/TrackerBetaTesting #361: Master section loudness management - very often going to clipping sound

Adam-Vandal, 2020-11-15:
Hi,
I work few days with Tracker, and almost alltime I have to pay attention to loudness management of tracks and master. I have still think on this, it is little bit annoying. and it distracts me from work on music.
Main thing is that it is very, very easy go to the clip with owful sound.
For example with Digitakt and similar machines it is almost impossible. (I can put overdrive to the sounds, volume to right, and it is still playing OK)
Can you make there some kind of bigger headroom, or simply some settings, that if I will go with all sounds to maximum, it will not go to that awful clipping with very bad sound?
Thank you guys!
Cheers

djkurzman, 2020-11-16:
Hello. I was getting ready to post a similar issue. Do you mean the general sample playback is being clipped?
In my experience, loud samples have been clipping very easily in 1.2.0 and up, forcing me to turn the instrument volume far too low to avoid distortion. But if I turn the actual sample volume down in the Sample Editor, say normalize to 60%, I am able to have the instrument volume at reasonable levels without clipping. Is this the same experience you are having?

Adam-Vandal, 2020-11-16:
Hi @djkurzman ,
yes, exactly.
Later I realized that little helps if I lower the main Volume (Master section / page 1), but I guess it should play normal if the volume is at 0 position and even more...

djkurzman, 2020-11-16:
If you lower the sample volume in the editor using the amplifier or normalizer effect to 50%, and then raise the volume back up in the Instrument Parameters or the master volume, do you still hear the sample clipping?

thedarkroomlondon, 2020-11-17:
Gain structure 101 here - if you've already got a bunch of samples all the way up to 0dB then there ain't no headroom left to sum them in.

Possible solution might be for the default sample/instrument volume to be set at -12dB rather than at unity.

prds100, 2020-11-19:
Hi

I second this issue, I received the tracker this week and I notice that the sound clips way too easily.  It seems to be primarily bass lines when Im applying resonance to the in built filter that pushes it in my experience.

vitodsk, 2020-11-20:
+1 Same here.
I posted it as new one, but I am having the exact same problem from 1.2.0 to 1.3.0b1, I get easily a lot of clipping. I also posted a video of this behavior in my posted issue.

vitodsk, 2020-11-20:
I hope that they really fix this as soon as possible. Is making the device almost unusable on my side. I don't know if you have the same problem of this video I am attaching.
[Uploading video_bug_volume_stutter.zip…]()

thedarkroomlondon, 2020-11-20:
@vitodsk you tried just turning the sample down more? 

The sum of two coherant signals is +6dB, so assuming that sample is normalised full scale, you'll need to trim it accordingly. Looking at your video, you've got it trimmed at -3.5dB so with two concurrent instances of that sample playing, you'll be clipping by +2.5dB. 

This is really a user issue, but it would be a convenient feature to have samples loaded with a default trim value, as well as control over the sample preview level. Setting both to defaulting at something like -12 or -18dB(FS), with an adjustment available in the config, would probably be a good starting point for those not familiar with basic gain structure principals.

djkurzman, 2020-11-20:
> I hope that they really fix this as soon as possible. Is making the device almost unusable on my side. I don't know if you have the same problem of this video I am attaching.
> [video_bug_volume_stutter.zip](https://github.com/polyend/TrackerIssues/files/5572510/video_bug_volume_stutter.zip)

I can't speak for the OP, but this is definitely not the kind of clipping I experienced. I was getting fairly consistent distortion on normalized samples unless I dropped the global or instrument volume very low. The only way I could fix it was to normalize the at around 50-60%, and then I could turn up the instrument volumes without the distortion. I believe it had something to do with the way the Overdrive send was being handled (even though I had it set to 0), and that's just an assumption on my part. That said, my issue seems to be fixed since I upgraded to 1.2.2.

To be entirely honest, the issue I'm hearing in your video almost sounds like a bad physical audio connection, like a failing cable or output jack.

thedarkroomlondon, 2020-11-20:
> I can't speak for the OP, but this is definitely not the kind of clipping I experienced. I was getting fairly consistent distortion on normalized samples unless I dropped the global or instrument [...]

He confirmed that in [this thread](https://github.com/polyend/TrackerIssues/issues/384) - using a TS instead of a TRS cable.

CV-Gate, 2020-12-07:
I'm also experiencing that clipping and also clicking when another chokes a hot sample. Probably more dramatic with lower frequency content.

PITmast, 2021-05-09:
Hi
When I’m playing for instance 1-shot sample with ideal start offset (I mean no clipping at beginning) It’s still possible to get a clip because of end which is completely random playing various note lengths.

I think, there should be kind of auto-micro fade out implementation with every dynamic cut of sample because of trigger

Regards

77377, 2021-10-01:
Yeees, managing and balancing loudness on the device is a bit of hassle atm.

danodayis, 2021-10-06:
just jumping on here (and created a github for this!) to echo this suggestion. many of us are using normalized samples from packs, which is fine, but that makes most of us having to turn our instrument volumes way down (even most of the demo tracks have their instrument volumes averaging at like -18db for each sample...).

the filter is a big offender here too. any strong amount of resonance will clip the sample and this seems to occur before the instrument volume so turning it down after the fact doesn't remove the clipping. I've found that the only work around here is to use the normalize effect on the sample at 50% or less, effectively halving the level of the sample, so it gives you some extra room to work with. if adding more per-instrument or master headroom is a hardware impossibility then i'd also suggest an option in the config that lets you import samples at a defined normalized level. Kontakt lets you do something like this where you can set the default sample level to be like -6db instead of at unity gain. An option in the config that displays something like IMPORT SAMPLE LEVEL: 1%-100% could set a default for all samples by just auto-normalizing them at the % level you want instead of manually processing each sample in this way

thedarkroomlondon, 2021-10-06:
@danodayis at the moment, it's working more like a traditional DAW mixer, and I feel keeping it that way makes a lot more sense to most users as it's a familiar starting point for any mixer.

An "import sample normalized to x dB" would however be ideal, with probably a default value of say, -12dB(FS). Percentage is a pain in the ass compared with absolute values (dB is log, how does the percentage scale against that?).

prds100, 2021-10-06:
@thedarkroomlondon yeah but this is an instrument meant for performance  not a DAW so I think that clipping and hot channels should be avoided at a software level as a general rule. While we use the tracker as a daw replacement, it needs to be able to perform in live rigs and if it’s so easy to clip that is a problem.

I’m currently testing the m8 tracker headless software. Also a hardware tracker (though with a different philosophy) and that handles volume infinitely better than the polyend. I hope this is addressed in future updates but the issue was mentioned very early in the trackers lifespan so I wonder if polyend feel the sound processing to be a feature rather than a bug….

thedarkroomlondon, 2021-10-06:
@prds100 I totally agree, that point was more addressing "I feel that 0db should be seen as a mid-point in instrument volumes" as I imagine that would cause some confusion with many users, as opposed to how the mixer is currently laid out. 

Absolutely agree that clipping should be avoided (unless intentional), and the easiest way to achieve this would be to normalize on import, rather than having to rebuild the entire gain structure downstream.

danodayis, 2021-10-06:
Oh jeez, yeah you're right - it does make way more sense to have the
"Normalize On Import" value be an actual ceiling dB versus a percentage.
That alone already confuses me!

Here's my 2c on instrument volume though: I think many people *are* using
it as a traditional DAW mixer, leaving it at 0dB, and then wondering why
their normalized samples are making their tracks so hot. Just like you said
it's a gain staging discussion. But to me it's almost as if the PT's
instrument volume seems to function as an input trim. For example: on a
traditional mixer you wouldn't have every track fader be down at -20dB but
when using normalized samples PT basically requires it to be that way (for
real, check out every demo project's instrument volume).

Since there's no way I've found to adjust input trim before Inst Volume
this leads to processing each sample via destructive Normalize or Gain
effects. But this would be extremely counter-intuitive to producers
importing (or recording) samples into the tracker, with a common
understanding of getting a healthy signal-to-noise ratio (i.e. - "big
waveform" to newer producers) only to have to turn it back down again.
That's very confusing for anyone that isn't already comfortable with signal
flow.