polyend/TrackerBetaTesting #364: [1.3.0b1] CUT and FAD not working

NickDimmock, 2020-11-16:
CUT and FAD are having no effect on audio samples in 1.3.0b1. To reproduce, simply add CUT or FAD to a playing sample. I've created a pattern on 1.3.0b1 with this issue, downgraded to 1.2.2 and had the same pattern behave as expected, and then upgraded again to 1.3.0.b1 and had the problem reappear. This was using the stock sample Legowelt/Bass/BASS-AiwaAcid.wav, but other samples have the same behaviour.

LetsSeeIfWeSolveThis, 2020-11-16:
Same here. OFF, FAD and CUT parameters share the same release time, which is the selected on volume envelope automation window.

danapeks, 2020-11-17:
Yep same here. Hope this one gets fixed asap.

closed 2020-11-18 by shuler323

doelman, 2020-11-20:
Indeed, same problem, makes 1.3.0 beta 1 unusable for me