polyend/TrackerBetaTesting #374: Lane Nudge UP/Down

phaelam, 2020-11-18:
would be nice for variations if a single lane could be nudged up or down

tfspsound, 2020-11-18:
Use insert to add/nudge cells down see my post right before this one for my request to remove/nudge up

phaelam, 2020-11-18:
> Use insert to add/nudge cells down see my post right before this one for my request to remove/nudge up

so I actually did a search before posting mine. sorry if it seems redundant. I looked at your post....and and is kinda different from what I'm requating. and would be a total pain, and different compared to what im asking for. you would mbe making changes to all lanes, and if you want to affect only one....that would just be an effort in frustration if you want to try things out id like it to werk like so....

original pattern...
[image]

select the lane or lanes you'd like to nudge with the "NUDGE TOOL"
[image]

Lane 1 got NUDGED down 4 steps, while lanes 2 and 3 stay the same. nudge would wrap around the pattern length.
[image]

tfspsound, 2020-11-18:
Sorry if you feel like my suggestion would be a total pain but you are wrong that it would effect all the tracks. Pressing Insert currently adds  a blank step on the track you are on this nudges the track down one step. Press it 3 more times and you’ve moved it 4 spaces down. That’s how it works (1.2.1) now. All I suggested which it seemed you wanted too was a way to delete a step, moving the steps up. I like the part about the wrap around think that’d be helpful.

phaelam, 2020-11-18:
> but you are wrong that it would effect all the tracks. Pressing Insert currently adds a blank step on the track you are on this nudges the track down one step.

OH right right...my mistake.i had done it while having steps selected across lanes.
but even doing that will kinda kill the flow. if you could simply nudge a track while it wraps around the pattern, you could try out stuff really quick. this is a request I keep making time and time again with AKAI. the MPC won't wrap. and if you have events on each step of a pattern, you can't nudge. 

I just love how easy it is to iterate on the Elektron boxes being able to nudge/wrap a track. 
but thanks for straightening me out RE: insert :)

lucasowens, 2020-11-19:
I came here to ask for this same thing. Basically a “rotate” feature from horizontal sequencer. The wrapping around part is the key.

tfspsound, 2020-11-19:
Now we’re wrapping, excuse the bad pun please. I started to think about your nudge suggestion too and think it fits with the invert function. So for example if you have a 32 step pattern and you select steps 4 to 12 then nudge them using a forward or backward command which would be in with the option to invert. The steps would then be nudged and wrapped within that selection only. 

Speaking of wrapping is there anyway of making the file menu wrap so when I press up at the top of the file loading list it goes to the bottom or pressing shift + home at the top takes you to the bottom. Also pressing left on track 1 on the pattern screen takes you to track 8.

phaelam, 2020-11-19:
seems like invert would change the sequence. its nice when a pattern within a pattern shifts slightly. makes for cool changes in a track. I hate this saying....but...."its the same but different"
and, an assumption only, I would imagine that nudging would be something doable while the pattern was playing. so you could hear the change immediately...and shift around till something strikes you.

but re: file wrap....YES!

phaelam, 2020-11-22:
can I add to this.

in SONG MODE, nudge would be great too. if you have an idea late in laying out your song....and want to stick a row at the top. it would be so much easier to grab a row and nudge it up to the top. instead of changing the pattern num and having to do that for all the following rows. I'm not the greatest at knowing what Im after right out the gate. and have ideas as I go.

unless there is already a way to insert a row at the top or mid song.