polyend/TrackerBetaTesting #375: Pattern in decimal start at 0-63 instead of 1-64?

neoj1n, 2020-11-18:
Is it possible to change the pattern decimal display to 0-63 instead of 1-64? That way, the Pattern Divider (4) would actually land on 0,4,8,12,16,20,24,28,32 like most trackers? Currently lands on 1,5,9,13,17,21,25,29,33, which breaks the 4/4 highlighting. I'm not sure if this is a bug or feature request, so I'll just place it as a feature request. :)

Currently using hex, since the divider for (4) is highlighting correctly.

dramaone, 2020-11-18:
Totally agree

SyntaxTerror, 2020-11-20:
Yes please!!

ketumatek, 2020-12-06:
Total agree ...easier like this especially if you are familiar to tracker

AmiDonk, 2021-03-06:
Yes

trutypesounds, 2021-06-12:
Yes please ☝🏼

marksumm, 2022-01-16:
Another victim of the CPU budget?

Jacknametrouble, 2022-01-16:
The option is available when hexadecimal numbers are used. We're not planning further configurability here as the amount of display options impacts realtime performance.

SandyMental, 2022-01-16:
you wont get a satisfying answer I bet :)

cant see the problem in telling the grid to start at 00 just via an option. 

I'm still hoping for somebody sitting somewhere silently and working on reverse engineering for a custom firmware :)
would be a total banger to have whole project around a custom firmware that is open sourced

Jacknametrouble, 2022-01-16:
Guys, Config >Hexadecimal numbers – On, Off > enable the throwback hex goodness. Pattern step numbers start with 00 and use sixteen digits (A means 10, B means 11, and so on all the way to F which means 15). This allows for easy alignment with music: bars will always start with pattern steps 00, 10, 20, 30, and so on. Each bar’s quarter-note beat steps end with the digits ..0, ..4, ..8, and ..C.

marksumm, 2022-01-16:
I know how hexadecimal works. This request was specifically about 0 offset with base 10.  Where did the idea that all early trackers used hexadecimal step numbering come from?  I'm watching a live stream of someone composing using Amiga Protracker as I type this and the step numbers are in decimal with 0 offset.

Jacknametrouble, 2022-01-16:
To simplify the workflow for non-hardcore classic tracker users, the Polyend Tracker is using a decimal system for values/data entering. Unlike most of the classic trackers, PT is displaying decimal values for the sequence step enumeration. You can change the sequence step enumeration to hexadecimal in the Config menu if you prefer so. We want the instrument to be accessible to everyone, even without the former classic tracker experience. Considering the number of users who have no issue with decimal and catch the flow, this works pretty well.

marksumm, 2022-01-16:
Again, this request is about decimal numbering with 0 offset.

trutypesounds, 2022-01-16:
> This is a genuine question: What is keeping a 180MHz ARM Cortex MCU so busy that it struggles with features available on a 7MHz Amiga 500? Is it the display? The increased sample rate?

i dont get it at all too, FT2 could handle 16-32 mono 16b44k tracks of polyphony + crazy complex envelopes, no probs with 00/HEX modes on crappy pentium1 60mhz... i get that polyend decided to use groovebox concept (pattern starts with 01) to attract more users who doesnt know anything about trackers .... well, ITS CALLED A TRACKER, make it a tracker

dramaone, 2022-01-16:
I just think starting at 0 and only having numbers is really sensible, for eg kick on 0, hat on 2, snare on 4. You can easily recognise exactly the placement you require in each pattern. On the Amiga with octamed is was always hats on 0,4,8,12,16,24, nice even numbers :)

Jacknametrouble, 2022-01-18:
> > This is a genuine question: What is keeping a 180MHz ARM Cortex MCU so busy that it struggles with features available on a 7MHz Amiga 500? Is it the display? The increased sample rate?
> 
> I dont get it at all too, FT2 could handle 16-32 mono 16b44k tracks of polyphony + crazy complex envelopes, no probs with 00/HEX modes on crappy pentium1 60mhz... I get that polyend decided to use the groovebox concept (pattern starts with 01) to attract more users who don't know anything about trackers .... well, IT'S CALLED A TRACKER, make it a tracker.

@trutypesounds We really appreciate your help, but would like to remind you https://polyend.com/manuals/tracker/#firmware-updates

If possible from the software implementation level. Polyend will fix any firmware-related issues considered as bugs. Polyend is always keen on hearing user feedback about possible functionality improvements. But is not anyhow obligated to bring such requests to life. We appreciate all the opinions, a lot, but cannot guarantee or promise their appliance. Please respect that.

Mike-w78, 2022-02-21:
I think all anyone is asking for is a switchable option to subtract 1 from each line number.  Surely that's not a biggie in terms of processing overhead?