polyend/TrackerBetaTesting #377: [1.3.0 b] Bug title

ilambert, 2020-11-18:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

bangcorrupt, 2020-11-18:
Duplicate of #364?

ilambert, 2020-11-19:
Yes. Didn't see that one.

closed 2020-11-19 by ilambert