polyend/TrackerBetaTesting #379: Please open source the firmware. Would love to help out with some issues (and also load in some custom DSP)

cmorgan, 2020-11-19:
Have you guys thought about open sourcing the firmware?

Would love to hack around on the Tracker. 
Keen to explore if it would be possible to add in custom effects into the audio pipeline.
Cheers for this awesome bit of tech!

vitodsk, 2020-11-20:
I asked the question about the firmware. Looks there are no plans to release it as open source.

CV-Gate, 2020-12-07:
I really second this. I think that it will not only benefit the users. Open sourcing a project like this can bump popularity dramatically. It also makes a difference between big companies' policies, which rarely will do. There are many stories of success, for example, Mutable Instruments.

eclectocrat, 2021-01-26:
Yes, I don't think that there is any competitive harm since trackers are not exactly secret sauce, your great triumph is the software and hardware together. I've written my own sequencers for Novation's Launchpad Pro (which is open source and one of the most popular live control surface on the market), and would love to fix bugs and contribute my own ideas to this wonderful piece of engineering.

closed 2021-02-16 by Jacknametrouble