polyend/TrackerBetaTesting #385: [1.2.1&1.2.2] Bug title

rothkane, 2020-11-20:
## Hi, sorry you had trouble!

### Steps to reproduce

I have just received a polyend tracker and a few of the demo songs that use a lot of reverb and usually 8 tracks playing at once, are crackling and popping on the headphones audio. If I turn the reverb down it stops or if I mute 1 or 2 tracks , is my tracker faulty? This occurs in both firmware versions 1.2.1 &1.2.2

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

vitodsk, 2020-11-20:
Hi, I reported the same, on the FB page. Basically you have to lower the reverb level. I am not sure if this is a bug or not.

rothkane, 2020-11-20:
Hi thanks for the reply , I was ready for sending it back , it's like the
CPU can't handle large reverbs with 8 tracks all running at once.

vitodsk, 2020-11-20:
To me happens even with only 1 sample playing if I use high reverb level.
I think the machine in general, or the latest firmware releases suffer of a big problem with levels. For example I am having A LOT of clipping and glitching with samples longer than 2-3 seconds at normal volumes. This is making the device pretty unusable.

rothkane, 2020-11-20:
Is this common with all the trackers ?  I have noticed the levels are too
high on all the demos seems like there is a lot of clipping going on. Which
Facebook group did you mention it on as there are two. Cheers Chris.

ambv, 2020-11-20:
The problem is this: there is a global setting controlling whether sends are pre-fader or post-fader. If the demo song was using a different setting from what you're using right now, it will produce clicks if the master levels are too high.

We will be fixing this by making the pre-fader/post-fader setting saved per project.

rothkane, 2020-11-21:
Thanks for the info I was a bit worried mine was damaged as it wasn't
handled very well during delivery and was loose in the box when it arrived!
Im currently midi syncing it up with my Amiga running octamed will gets vid
up soon. Cheers Chris.

Stromchaos, 2020-12-07:
Hi there, i got my Tracker this Weekend and have still the same problem. i have upgrade the Firmware to 1.2.2
Sometimes i still can hear it, especially in the Demo Song "Sam Slater".

matozoid, 2020-12-11:
Just received mine and have the same problem. It's not a great first experience :-(

zobbo, 2020-12-11:
For info it's happening with the most recent beta as well. Not come across
it in my meagre offerings but can certainly hear it in Sam Slater's
composition. When more is happening it gets worse.

Sounds like something is getting too busy maybe ?

1.3.0 beta 2

ambv, 2021-01-05:
The default setting for Sends mode is now Post-fader in 1.3 so this is fixed.

closed 2021-01-05 by ambv