polyend/TrackerBetaTesting #388: [1.3.0b] I change the instrument of steps but, when I change of pattern, the change does not save

LetsSeeIfWeSolveThis, 2020-11-20:
I don't mean about the classic project save function [File], nope! 
I mean Tracker is not saving instrument changes on pattern edit mode. When you change of pattern, you loose the changes (only for instrument column)

ambv, 2020-11-20:
I cannot reproduce this. Can you share your project and record a short video demonstrating this?

LetsSeeIfWeSolveThis, 2020-11-21:
This behavior is different from the described, but is interesting; **I change of instrument while the pattern is reproducing and, when the pattern ends, the value of the instrument resets and go back to 1 again** (staying on the same pattern), note; this only happens in red mode --> https://drive.google.com/file/d/12lPd1skUs8dDJP8cEwe3Kztg_TnVugIv/view?usp=sharing

BUT on green mode, if I change the pattern and go back to the original, the instrument wont save. --> https://drive.google.com/file/d/12jDDvgH_y3DRfgasbVRubQILn-lAt9kG/view?usp=sharing

---
Sorry, I can't do noises right now. Please watch the video in HD to see the numbers better.

LetsSeeIfWeSolveThis, 2020-11-21:
Should I share the project you see on the video anyway?

LetsSeeIfWeSolveThis, 2020-11-21:
OKAY I found I new possible way to reproduce it;

It only happen when I **change the instrument with the knob**, as you see on the next video, I tried to change an entire column of instruments directly with the pads and this issue does not appears;
https://drive.google.com/file/d/13DAJp7Kupzyh7Njl8eap4ghIZ9MBXyJY/view?usp=sharing

The project that you see on the video is exactly this one; 
https://drive.google.com/drive/folders/19E3PhppduVoGk8f-8dmSzQ31VOp8gu7E?usp=sharing

LetsSeeIfWeSolveThis, 2020-11-21:
And this behavior is different from the described, but is interesting; **I change of instrument while the pattern is reproducing and, when the pattern ends, the value of the instrument resets and go back to 1 again** (staying on the same pattern), note; this only happens in red mode --> https://drive.google.com/file/d/12lPd1skUs8dDJP8cEwe3Kztg_TnVugIv/view?usp=sharing

This behavior is not happening all the time, just yesterday, but was really weird and related in someway with the bug itself, so I wanted to mention it instantly.

---
Sorry, I could not do noises last night. Please watch the video in HD to see the numbers better.

guiguigui77, 2020-11-21:
hello,

same here, but even when i change an instrument with a pad (in rec mode) it doesn't save the change when i move to an other pattern

guiguigui77, 2020-11-21:
i also found that copying that pattern to a new one makes the change saved, in the new one at least

LetsSeeIfWeSolveThis, 2020-11-21:
Hi, sad to hear that, I'll write here if it happens to me also with pads. I just notice it today, I didn't try it yesterday sadly. Maybe it was just a casual and sporadic fix.

And about your second comment, yes!! I noticed it too, thanks for mentioning it, I forgot to do it!

BerndPeuser, 2020-11-24:
I can confirm that bug too on firmware 1.3.0b.  Thank you @guiguigui77 for the tip to fix it.

closed 2020-11-24 by shuler323