polyend/TrackerBetaTesting #392: Add / Add Next in Sample Loader

vitodsk, 2020-11-21:
Hello,
in Sample Loader seems to me that Add and Add next behave in the same way, if you have the cursor on one instrument on the right, using "Add next" does not try to add the sound to the next slot, but in the same slot you are pointing at.

I don't know if this was the original purpose of "Add next", but I wondered why having 2 operations that do the same thing ? So I think "Add next" was intended to add the instrument on the next slot, but that is not what's happening.

Is this intended ?

meteorist, 2020-11-29:
I had the same experience however I noticed that Add Next will not only overwrite the currently selected sample slot but also jump to the next slot afterwards.

It could possibly have been designed this way but from a UX standpoint it's not very intuitive. It would feel a lot more natural if it jumped to the next slot first and then add the sample afterwards.

gherat, 2020-12-05:
O agree add next should add the sample to the next empty slot.

ketumatek, 2020-12-06:
That is true...no difference between add and add next...
But normally add next should be add a sample to the next slop..
May be it will be like that in the futur...

pinkcell-niico, 2021-04-02:
Add Next : adds sample in cursor location (same as Add behavior) AND then moves the cursor to next slot.

But definitely, Add Next should become **ADD TO NEXT EMPTY SLOT** (and moves cursor to that slot so we can see the number of that slot)

bambrose, 2021-04-02:
Yeah that makes sense to me

closed 2021-12-02 by Jacknametrouble