polyend/TrackerBetaTesting #395: Pad sensitivity issue

koraysels, 2020-11-21:
Anyone else having issues with pads not being consistently triggered when pressed?

Ik makes finger drumming quite a challenge..

deixic, 2020-11-21:
Yes, very inconsistent here too.

LetsSeeIfWeSolveThis, 2020-11-21:
I can't play two pads at the same time. Like a monophonic finger drumming experience.
Can you confirm this?

koraysels, 2020-11-22:
Okay that solves your polyphony issue but I really have a problem even on an emtpy pattern first track. The pads are really "stiff" and do not consistently register finger presses. 

Could this possibly be fixed with software?

LetsSeeIfWeSolveThis, 2020-11-22:
I have to give you the reason anyway, I mean, I always noticed issues in finger drumming, beyond my last comment issue.

But being too noob with this device yet, and thinking about the polyphony limit... I can't assure nothing yet. Maybe the release times of some drum samples when I play faster make it fails, or maybe its a real issue and something fixable thru the software. I need more time to verify, but It don't seems to be a hardware issue.

koraysels, 2020-11-22:
Sorry @LetsSeeIfWeSolveThis but maybe you can open a seperate issue? I am actually talking about pad presses not registering.. even just a single pad. Nothing to do with polyfony, please open a seperate issue if you have other issues.

LetsSeeIfWeSolveThis, 2020-11-22:
''_even just a single pad._'' I was just confirming If we had the same problem, you didn't mention that, please specify your problems before posting them.

koraysels, 2020-11-22:
Okay cool, so you are a contributor? What i mean is like I said originally. I have a problem with overall consistency of the pad sensitivity for registering presses. With my unit it is not consistent;) apparently others have this too?

Just leaving it out here in case the software makers come across this and think they have a solution. Have a nice day.

deixic, 2020-11-23:
koraysels: I have the exact same problem as you, inconsistent pad sensitivity.

math0ne, 2020-11-25:
I experience this for sure, especially noticeable when entering hi hats into a beat.  If I use an external midi controller with pads, it's perfect.

koraysels, 2020-12-26:
Damn I dont want to have to use an external midi controller. 

I wonder if the new hardware revision fixes this and if so, if we could send in our units for the new ones (rounded corder buttons version)?

visualearner, 2021-04-06:
I have the same issue. It makes entering notes in time nearly impossible which when making music is a bad thing. Please fix.

x2db, 2021-04-19:
> Damn I dont want to have to use an external midi controller.
> 
> I wonder if the new hardware revision fixes this and if so, if we could send in our units for the new ones (rounded corder buttons version)?

A new hardware revision ? could you share link where this was discussed ?

closed 2021-12-02 by Jacknametrouble

szszoke, 2022-01-01:
I am having this issue as well with a second hand unit.

The unit was purchased from Thomann at the end of November 2021.

The pads require a rarher strong tap to trigger. All of them.

szszoke, 2022-01-01:
> Damn I dont want to have to use an external midi controller. 
> 
> I wonder if the new hardware revision fixes this and if so, if we could send in our units for the new ones (rounded corder buttons version)? 

I seem to have the newer revision and the pads also feel a bit hard to trigger for me.