polyend/TrackerBetaTesting #398: Lane lock request

manksy, 2020-11-23:
I am enjoying my tracker, however there is one thing that would make it infinitely more usable for me, the ability to lock a lane and prevent
spill-over to the next lane, meaning notes on a repeated step would overwrite what is there.
I totally get the usefulness of the spill over for drums and rhythmic content, however for melodic content to me I want to stay kind of “per track” and having notes from a bass line go into other tracks just kind of distracts and lengthens my workflow by quite a bit.

tfspsound, 2020-11-24:
I think this would be a useful option to have.

ROM72, 2020-11-25:
Yes, I completely agree on that. Wanted to ask the same.

BverseJohn323, 2021-06-20:
Please make this happen!