polyend/TrackerBetaTesting #401: [1.3.0b1] Changes in pattern mode get erased in song mode

dustinpero, 2020-11-23:
I've noticed when I try to add a pattern on instrument 10 track 8 with a pattern that already exists in song mode (pattern 7). My changes get erased after playing up to pattern 7 in song mode. 

* How often does this bug happen? 
Every time.

* Does this bug happen in previous firmware versions?
not that I'm aware of.

* Does this bug happen in every project?
Yes, i loaded a new project and tested it with a single sample on instrument 10. Pattern 1 in track 8 works as intended, but once I switched from pattern 1 to 7 in song mode, my pattern was erased again in track 8 on pattern 7.

BerndPeuser, 2020-11-24:
https://github.com/polyend/TrackerIssues/issues/388

shuler323, 2020-12-10:
> #388

:)

closed 2020-12-10 by shuler323