polyend/TrackerBetaTesting #403: [Firmware version] Two notes playing when replacing one

Konevitsaus, 2020-11-24:
The newest official firmware still has one thing that really messes with my creative flow: when overwriting some existing notes with new ones, it very often plays the replaceable note as well, not only the replacing one. Highly irritating at least for me. Any chance to fix this?

shuler323, 2020-11-24:
Please add some more info - at least FW version where this bug occurs and steps to reproduce

ambv, 2020-11-25:
Ideally we need a short video that shows the issue.

Konevitsaus, 2020-11-26:
I hope this got uploaded. There's a video about me playing one note and then overwriting it with another. You can hear how at one point the old note is repeated between the new ones.

This one was quite a quick one and featured quite a little annoyance. But sometimes when tracking, it's almost constant. Very annoying.

[trackerbug.zip](https://github.com/polyend/TrackerIssues/files/5604790/trackerbug.zip)

Konevitsaus, 2020-11-26:
(My firmware version is the latest one – I mean, even as a beta.)

shuler323, 2020-11-27:
thanks for video, we'll work on it

Konevitsaus, 2020-12-26:
> thanks for video, we'll work on it

Actually, now I noticed that it mostly happens when there's a note right next to the one I'm replacing – that NEXT note starts to play at the same time. Very, very distracting when trying to keep some new melodies in my head and put them on the track.

closed 2021-01-29 by shuler323