polyend/TrackerBetaTesting #408: [1.2.2] Sample Naming Improvements + Copy/Paste Names Feature

Zombie-Joe, 2020-11-24:
Naming samples is not fun in the current firmware [v.1.2.2]. If possible, can you please add in the future the ability to have the name field remember the last saved name when you go to save a newly recorded sample? For example, I've recorded a Kick from my Drumbrute Impact as "DRUMBRUTE IMPACT Kick 01". When I adjust my Impact's knobs to change the tone and record a new sample, I have to then type "DRUMBRUTE IMPACT Kick 02" all over again to maintain my naming convention. This is very time consuming and the lighted keypads are not very friendly to type on.

The ability to highlight the text (by holding shift and using the left/right arrow keys) in the name field and being able use the Copy/Paste buttons would also be very helpful.

Personal Note:

I would also like to say that I love my Tracker! I have been spending about 3 hours a day lately sampling all of my gear into my Tracker (Behringer RD-8, TD-3, Arturia Drumbrute Impact, Microfreak, Minibrute 2S, Moog Sub Phatty). Everything is sampling great and I love how simple samples are recorded and how easy they are to trim. I have recorded 100s of samples directly into my Tracker, even with long names like the above example :+) I want to thank you for making such an easy to use, focused, and fun way to produce music!

ghellings, 2020-11-24:
Actually, autocomplete for naming would be cool, populated by common prefixes of existing samples.

Zombie-Joe, 2020-11-25:
> Actually, autocomplete for naming would be cool, populated by common prefixes of existing samples.

That would be awesome too! With your idea, that makes 3 ways they could improve naming samples.

closed 2021-12-02 by Jacknametrouble