polyend/TrackerBetaTesting #409: [1.2.2] Screensaver feature request

Zombie-Joe, 2020-11-24:
When I sample for hours, sometimes I may need to get up and stretch, get food, etc. which leaves my Tracker on the same screen for a long time until I come back. Can we please have a screensaver to protect this amazing machine from image burn? OFF, 2 min, 5 min, 10 min, 15 min, timer options?

Can we also customize this screensaver? ;+) Maybe we could import images to be able to use for a screensaver? Think Windows 95 style screensavers, doesn't have to be too fancy. There could be a folder on the SD just dedicated to this feature where you would place your photos (sized to the correct resolution/bits/color depth of course). Thanks!

tfspsound, 2020-11-24:
I like it

bgdnrz, 2020-12-28:
+1 for totally black screensaver.