polyend/TrackerBetaTesting #413: Performance mode breaks Song to loop Pattern

tfspsound, 2020-11-24:
‘When switching from the Song mode to Perform mode, the currently played pattern is looped’

Can we have the option to not loop the current pattern as well? I can see the benefits of the chain being broken and looped for improvisation but having to swap to back to song mode to advance the song breaks performance flow cause I can’t apply a Performance effect whilst I’m on the Song screen. So the option to not loop the current pattern would be nice I think.

anemlife, 2020-11-30:
100% agree! That would make the performance mode way more usable to do recordings of a final edit for example.

tfspsound, 2020-12-01:
Exactly. I was also thinking if they could implement this option then create a Performance Fx called Pattern, then we could use to that to navigate the song by setting up 3 cue points to jump between.

ketumatek, 2020-12-06:
I ask the same request...may be I will close mine ...
Loop mode should be only a choice when you are in perform mode..
Hope they will change that.

VincenzoUser, 2020-12-19:
As I expressed in replying to a more recent post on the same topic, I totally adhere to this request... Thanks!

ambv, 2021-01-19:
Performance mode loops the current pattern because you can swap out tracks with tracks from other patterns. It's unclear what should happen to those swapped out tracks when a pattern would change because the song kept playing. It's unclear what should happen when the new patterns length changes but there's already an effect in place to change pattern length. And so on, and so on. The intended workflow so far is to perform on a single pattern and continue song from the Song screen.

Let's keep this open to discuss a possible design for this feature. Keep in mind that the behavior of looping the current pattern would need to be somehow preserved as well because most users seem to be fine with it.

ambv, 2021-01-19:
(closed by accident)

ketumatek, 2021-01-19:
> Performance mode loops the current pattern because you can swap out tracks with tracks from other patterns. It's unclear what should happen to those swapped out tracks when a pattern would change [...]

May be you can make as option as if we want or not the loop mode in performance mode...
Like this users who like in that way stay like this 
and for those who don't want the loop pattern in performance mode they can switch off this possibility...
May be in Config / General : you can put this like
Loop mode in performance mode            on or off

tfspsound, 2021-01-19:
Hey. Thanks for looking into this. Not sure if I agree that most user would like to keep it the way it is I think the vast majority of users aren’t represented here either they won’t want to join GitHub or just can’t be bothered, or maybe they just use the tracker for a specific feature so they don’t encounter the problem. I use the tracker for programming beats primarily which is why this option is so important for me. I’d be happy for a config option like ketumatek suggests above , Perfomance Play :  Pattern  / Song so it either loops the pattern as it does now or applies to the whole song, or as extra options on Pattern Play Mode like Loop but this would mean changing it as is now to play thru the song in Performance Mode allowing the user to choose which pattern to loop as it’s playing. 
I don’t use the Song screen much at all, I particularly dont copy and paste tracks from one pattern to another as I found it kind of buggy, (it pasted the track cell I copied onto other patterns in that track with a space between them?) I also found it difficult to figure out which block is which so I tend to stay away. Copy and Paste Pattern or just copying the track from one pattern to another works fine for me although if you can get copy and paste to work in song mode that’d be cool. 
If the pattern lengths don’t match how is it such a problem? Surely the pattern lengths will be some form of multiples of each other eg 8 steps, 16, 32, 64, 128 for 4/4 so wouldn’t it just be a case of repeating the pattern to fill out the length of the longest track? If somebody is trying to put a 13 step track onto a 27 step track then I guess that’s up to them but maybe the tracker isnt the right machine for them as, as far as I understand the tracker can’t do complex polyrhythms of that nature. 
I love Performance Mode it’s the icing on the cake but like I’ve said before we perform songs not patterns. Thanks again for the hard work you guys are putting in at polyend it’s worth it and I’m sure I’m safe in saying we all appreciate it. Cheers

anemlife, 2021-01-20:
I agree about an option to toggle pattern looping on/off in the config Menu! This would allow to keep the old workflow + a new way for using the Tracker (which is the one I'm looking for also!)

Thanks! :D

VincenzoUser, 2021-01-20:
> Performance mode loops the current pattern because you can swap out tracks with tracks from other patterns. It's unclear what should happen to those swapped out tracks when a pattern would change [...]

@ambv First of all, thank you so much for your willingness to listen to the needs of users and the commitment to implement the (already very GREAT and INNOVATIVE) potential of the Tracker!

I try to simplify the problem.
Let's imagine an artist playing very actively in Performance Mode on a song pattern: his hands are very busy and the performance is very "scenic".
Suddenly, to keep the song going on, he switches to Song Mode and the song goes on: his hands stop, the performance becomes passive and the music has almost another style (more automated, like on a DAW).
If the artist's hands could do something to manipulate the sounds even in Song Mode, that would be great!

The solution proposed by @ketumatek would be great: the possibility to choose between the two options.
Even more, I would love to decide on the fly during the performance whether to stay on one pattern or continue the song (the point is that in both cases I would like to be able to play with the effects in real time).
But from your words I understand that there is another stumbling block: it's unclear what should happen when one or more Performance Mode effects are applied to a pattern when that pattern switches to another pattern (different leght, different elements, different sounds) because the song kept playing.
My answer is: nothing, in my opinion. It should simply remain the effect applied to the previous pattern even when the next pattern arrives, until the artist's fingers turn off the effect. Just like in the Octatrack, Deluge and any other similar device.
Just as an example: if I press the Deluge stutter on a song pattern and hold it while the pattern changes, I won't even hear the pattern change but when I take my finger off the stutter I will find myself at a more advanced point of the song. Or instead... if I increase the reverb, I hear both the first and second patterns reverberated. It depends on the effect, but in any case the effect is not an obstacle to the continuation of the song.
The same thing you can see in the stutter and repeat effect or reverb
 (or any other effect) of the Octatrack or any other similar device.

But there is an easy experiment that shows how it should be in the Tracker as well.
Create three patterns of different lengths, for example:
1) 32 - that we call PATTERN A
2) 64 - that we call PATTERN B
3) 96 - that we call PATTERN C (that is exactly the sum of A + B)

What should happen if you play with effects in Performance Mode when switching from PATTERN A to PATTERN B?
Simple: exactly what happens if you play with effects in Performance Mode for the entire duration of PATTERN C.

What do you think about?
Thanks again for everything!

tfspsound, 2021-01-21:
Found a few more request for this 
#322 
#179
#164

ketumatek, 2021-01-21:
> @ambv First of all, thank you so much for your willingness to listen to the needs of users and the commitment to implement the (already very GREAT and INNOVATIVE) potential of the Tracker!
>
>
>
> I try [...]

Really nice idea...I proposed to switch on or off the pattern in performance mode because normally (for me) when you perform your tracks you know it and you know what will arrive in the next track...
Plus in performance mode the fx you are using are just in a small moment to put more stuff and change to your tracks ...like a remix track...(it can be a lot but really short)
Of course if you put a lot of fx in your track you have to manage with them during your performance...

But I think that, if you know that you will perform your track you won't put to much effect or you will use some fx and not some other ...
And for me, it is easier if my track is like in song mode to perform on it more than in track loop mode..
And if i want to loop just in a part of my song i always have the loop mode that I can use when you want...
For me it's more clear to put the loop mode when i want, in performance mode, more than shout off each time i want to play the next pattern of my song..
More easy for the work flow ...and live performance...

Hope it make sens...

ketumatek, 2021-01-21:
> Found a few more request for this 
> 
> #322 
> 
> #179
> 
> #164 

Nice ...in the firmware 1.1 (#332) it was possible to play the song entire in performance mode...
Hope they will go back to this with an option as I explain before ...
As this every one will be happy ...those who want to performance on one track and those who like to perform on the all track...

tfspsound, 2021-01-22:
So I was havin a play today and I think this could be done quite easily without anything changing on the surface. 
If as suggested some kind of Performance Play Mode : Song / Pattern is added to the Config options then all we would require are a couple of navigation options added to Pattern Play Mode. Currently we have FWD / REV / RND / C1-10 but if you could add 

LOP (loop the current pattern only available if the config option is set to Song) 
LPR (loop the current pattern in reverse again only available in song mode). 

Then if Pattern is selected in config, options 

NXT (play next pattern) 
PRV (previous)
00 that can be changed to jump to the matching pattern number making cue points. 

How the Pattern Play Mode FX settings work would depend again on the config settings so if set to Song FWD plays forward through the song and REV would play backward not only through the pattern but the sequence of patterns in the song too 5,4,3,2,1.... RND could remain largely as is being random, but the song would still play forwards. 
With the config option set to Pattern the functions of FWD / REV .....would remain unchanged. 

As VincenzoUser says leave the fx as they are when it changes from pattern to pattern whichever mode they are in, that’s the concern of the performer to craft but I think these suggestions would make navigating about patterns whilst remaining on the Perfomance Mode screen alone much easier and help us maintain flow and add variation to our compositions. 
Cheers

ambv, 2021-04-26:
Fixed in 1.4.0b1.

closed 2021-04-26 by ambv

tfspsound, 2021-04-30:
👍