polyend/TrackerBetaTesting #415: [1.3.01] Song stems are not in time with each other when exported.

Stazma, 2020-11-25:
Hi there,

So i've been trying to export a few songs as stems to work on the mixdown in the computer and a old bug seem to have resurfaced, the stems are not perfectly in line with each other, some seems to drift a little with time so the more the song advance the more elements gets unsynced
[image]

* It seem to happen with every tracks I tried to export

* I remember I did mention this before but I also remember it being fixed at some point.

doelman, 2020-11-25:
Same as #281 I don't think this has been solved in 1.3.01 as this bug is still open?

shuler323, 2020-12-07:
Export function will be implemented in new way, you can expect fixing your issue in some new betas, which will be noted here.
(#281)

closed 2020-12-10 by shuler323

dfordabs, 2022-06-09:
firmare 1.5.0 - still not fixed