polyend/TrackerBetaTesting #422: 1.2.2] Equalizer not working

thebradclay, 2020-11-26:
### Steps to reproduce

Place a sample in the editor and use the eq. When trying to cut, low mid and high. There is no audible difference when pressing "preview: or after hitting "apply". I have tried to cut with multiple samples and no results. When boosting the low there is an audible difference when hitting "preview" and using "apply". 

### Attachments and links

I will try to get a video uploaded this weekend.

### Other important information

* How often does this bug happen? every time

* Does this bug happen in previous firmware versions? I don't know

* Does this bug happen in every project? Yes

wjakobczyk, 2020-12-14:
The effect of cutting can be subtle, especially when compared to boosting, depending on the sample, but it is there. Try also tweaking the freq.

thebradclay, 2020-12-14:
> The effect of cutting can be subtle, especially when compared to boosting, depending on the sample, but it is there. Try also tweaking the freq.

Subtlety is of no help when I am trying to make cuts to a frequency.

dbass327, 2020-12-18:
I’m having this same issue and got told to put a bug report in. I didn’t want to dial report so I’ll start by commenting and if I need to do a separate one I can. Firmware v 1.2.2, and the equalizer in the sample editor has no audible effect even at -12 db. I shot a video replicating this I’ll include a google drive link to. After the video I tried other effects and they worked fine, so I know I had the sample selected in the editor. 
 https://drive.google.com/file/d/1jJwNZZfUFzBRKMAndNFqRscSTM7WDZpk/view?usp=drivesdk

thebradclay, 2020-12-18:
> I’m having this same issue and got told to put a bug report in. I didn’t want to dial report so I’ll start by commenting and if I need to do a separate one I can. Firmware v 1.2.2, and the equalizer [...]

Maybe submit a report. I have tried the eq on multiple samples and it's like it just doesn't do anything. I've also tried the eq in the v1.3.0b2 and v.1.3.0b3 and it's the same issue. Unless I am totally overlooking how to use this specific eq, which could be possible I guess, but I think it's not working properly.

trutypesounds, 2022-01-18:
dont close this one, the EQ must be done in the right way

shawnroos, 2022-02-21:
Thank god this is a bug, I thought I was going crazy 😬

trutypesounds, 2022-07-04:
@MortalJS is this still in priority??

closed 2023-04-21 by MiroPoly