polyend/TrackerBetaTesting #423: [1.3.0b1] Sample rendering bug when switching patterns

mroemore, 2020-11-26:
If you select all steps on a 64 length track, then use the Pattern button + jog wheel to switch to a pattern with 32 length, then 'render selection', you will have a sample of 64 length.

### Steps to reproduce

1. Create pattern A of length X.
2. Create pattern B of length Y where Y < X.
3. Select X notes on pattern A.
4. Use Pattern button and jog wheel to switch to pattern B.
5. Use the 'Render Selection' functionality.
6. The resulting sample will be of length X, with an X-Y length of silence at the end.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Happens every time.

* Does this bug happen in previous firmware versions?
Probably, but not sure.

* Does this bug happen in every project?
Yes.

mroemore, 2020-11-26:
I've just noticed something more strange... after experiencing this bug it is still rendering samples that are 64 steps long, even after de-selecting and re-selecting the steps, in a pattern of length 32. I will try to upload a video of this soon.

shuler323, 2021-01-06:
i've reproduced already, no need for extra video, fix will be soon, thanks for reporting :+1:

closed 2021-01-07 by shuler323