polyend/TrackerBetaTesting #426: A ”copy / past on next pattern” Screen key

PaulAntioche, 2020-11-27:
in pattern mode
It would be very interesting a Screen key which allows to copy + paste a pattern on the next empty pattern in a single click.
The functions "copy pattern" and "paste patter" are already existing with hold Pattern key + Copy key> go to new Pattern and hold Pattern key + Shift + Copy (Paste) to paste…
So, it's possible to replace the ”copy pattern” and ”paste patter” Screen keys by just one ”copy / past on next pattern” Screen key.
A Screen key like this would be incredibly useful: If you are on pattern 1 and you press the ”copy / past on next pattern” Screen key, you will find yourself on pattern 2 which is the copy of pattern 1 which allows you to evolve the composition without cutting the flow.

wesleyhodgson, 2021-09-11:
YES! This is a big part of why I love the Monome sequencer Kria: being able to easily jam on new patterns without dead-air while I fumble for the paste buttons. :)

Monsieur36, 2022-07-09:
Relatively new to Tracker, so please accept my apologies if this is now possible and I've missed it.

If it's not yet possible (i.e., with firmware 1.6.0), I would also very much like to be able to have what I'd call a "proper" Duplicate Pattern function - even just current Pattern to next (empty) Pattern slot, but certainly without stopping or interrupting playback. :)