polyend/TrackerBetaTesting #428: Colors for each track in track mode and song mode

ketumatek, 2020-11-27:
Hi

It will be very nice if we can chose colors 8 or more for each tracks ...we can rename each track but with a different colors for each track it will be easier to know what is in the track. 
Ex: drum in red orange yellow
       Bass purple
       Synth green blue 
       Etc
Colors that you can find in song mode.
like this we can choose a different color for each slot in song mode...
So song mode will be more clear and visual. It is not only blocks with the same white color..
May be you can use some 8 simple colors (one for each track) or custom color that you can apply in fonction of your love.
Hope it make sense

Adam-Vandal, 2021-01-10:
Hi, 
wanted report this thing, but realized that it is already reported :)
yes yes, it often happened to me that I mistakenly edit the wrong track. If there could be a possibility to collor tracks in Pattern mode, this will never happen anymore (I hope :D )

ketumatek, 2021-01-10:
> Hi, 
> 
> wanted report this thing, but realized that it is already reported :)
> 
> yes yes, it often happened to me that I mistakenly edit the wrong track. If there could be a possibility to collor tracks in Pattern mode, this will never happen anymore (I hope :D )

Unfortunately I read somewhere that it won t be soon...but I hope they will seriously think about...will be a must for me..

dias236, 2021-09-02:
It would be Nice and Easier to use the song mode with color coded tracks, where patterns have distinct tones of their track color.

dias236, 2021-09-02:
It also would be super dope to have a small number identifying the pattern on the song mode rectangles

65535a, 2024-08-03:
Just got my unit yesterday and I love it. The lack of identifiers for different tracks in song mode is the only real downside I have encountered so far.