polyend/TrackerBetaTesting #432: [1.3 beta] Midi In Arturia Keystep sound attenuation and Midi adapters. Issues ?

Brulur, 2020-11-28:
Hi, is there a way to cancel velocity on Arturia Keystep ? When I play notes in Playback mode the sample sounds louder than in pattern mode. I already tried to switch 'Only notes' and 'Velocity' in parameters and I switched what should be in the Arturia Midi Control. I also reinstall firmware 1.2 but nothing changes. 

I also bought two midi adapters as attached. The only thing that works is when I use the Polyend adapter for midi in or out directly plugged in the tracker. The two others doesn't when they're directly plugged in the tracker. Funny thing, one of them works (!!!) when it is not directly plugged in the tracker but in my 0 coast (and the polyend adapter in the tracker) but not the other. Both work well when directly plugged in my 0 coast and the Keystep. 

Thank you in advance for your answers.
Yann 

[image]

konni-arcadia, 2020-12-02:
Hi, my Korg midi-trs adapter doesn't work either. The adapter shipped with Tracker work fine.

Can this be corrected via firmware?
In the meantime is their a list of confirmed working adapter?

benc-uk, 2020-12-18:
There are two types of MIDI to mini-jack adapters. Type-A and Type-B, and they are fundamentally incompatible, so you need the right type of adapter for each device.

See:  
"A simplified guide to MIDI over TRS minijacks – minimidi.world" https://minimidi.world

Plieuw, 2021-01-02:
It would really help if u could change the type a or b, in the config.

Type a will become the industry standard. So in order to make the tracker future compatible in small setups, please...

benc-uk, 2021-01-02:
I expect this is not something that if configurable in software otherwise every device that uses TSR-MIDI would have an option for this

ambv, 2021-01-17:
@benc-uk is right, there isn't anything we can do here software-wise. The device's hardware expects MIDI TRS Type B. It's documented.

closed 2021-01-17 by ambv