polyend/TrackerBetaTesting #433: Tails on/off option for render/export

keith-macleod, 2020-11-29:
Hi, I find myself often having to approximate a crop length for a sample that I've exported, e.g. a beat, so I can then equal slice and mess with it further. It would be nice to have the option to enable/disable tail on a per-case basis.

gherat, 2020-12-05:
Yes please!!! We really need this to accurately re sample and slice in equal slices

gherkins, 2021-07-04:
+1 
Having the exact bars length at least in the pattern stems, would make importing (in e.g. ableton live) so much easier!

gherkins, 2021-07-04:
Ableton Live has the "Render as Loop"-export option.

When checked, the FX tails for the selected export region will be included in the beginning of the file, rather then appended of clipped. 

Optional append would still be cool,  but maybe that is an even better solution.

gherat, 2021-07-04:
Someone had a hack for rendering loops, haven’t tried it but he said it worked. So it goes as follows: Make your sequence 1 step longer, then put an “OFF” parameter on that last step, then render the selection of the original sequence. That way it will be forced to cut of the tail and we will be able to finally resample a sequence that we can “cut into equal slices” Basic functionality if you ask me. 

Sometimes I think there are only programmers working there and no musicians. This thing has so much potential but 80% of the tools are just programmed wrong. IE: gate times, slices won’t play through, volume lfo’s that just don’t play through etc etc etc.

gherkins, 2021-07-06:
👍 nice workaround, thx for sharing!

> Sometimes I think there are only programmers working there and no musicians

As a programmer and musician I highly doubt that. 😅   

Personally I think being able to file issues on Github about your favorite new features  - for a musical instrument - is really cool. Also the firmware is pretty awesome for being a v1.    

That being said, being able to export stems in the exact bars, would be a really nice thing to have!