polyend/TrackerBetaTesting #437: Beat slice not pads efficiently

tfspsound, 2020-11-29:
Steps to reproduce

Get a long sample and beat slice it so you have over 48 slices. The slices are arranged from the 2nd row up so there are only slots for 36 samples on the remaining keys. 
The bottom row of pads is filled with duplicates of the first slice so the only way to access them is by increasing the Note value on the Pattern screen. 
Probably not an issue if your using an external keyboard but if you use the pads it’s difficult. 
Would it be possible to use the Slice button to preview the slice so that if this gets rectified it will be possible to preview slices if you make more than 48 and aren’t using an external keyboard?

### Other important information

* How often does this bug happen? 

Every time

* Does this bug happen in previous firmware versions?

1.2.1

* Does this bug happen in every project?

Yes

modulationsource, 2020-11-29:
The slices are connected to notes, not to pads.
So, if you change the root key in the config page, you can use all 48 pads for the slices.

tfspsound, 2020-11-29:
Nice one. Cheers

closed 2020-11-29 by tfspsound