polyend/TrackerBetaTesting #438: Make Step, Step all the time

tfspsound, 2020-11-29:
I’d like it if Step applied to the arrow keys so that if Step is set to 4 and I press down the cursor moves down 4 cells. Since we don’t have a CTRL button maybe holding Shift and pressing down could select every fourth cell, making it easier to extract something from a track. If this interferes with people’s current workflow how about holding Step?

muhfugga, 2020-11-29:
I like this idea

keith-macleod, 2020-11-29:
This would also make it much less painful to edit long patterns.

tfspsound, 2021-01-28:
I made amendments to this idea on other post that I have subsequently lost but in brief. 
Make the Step screen key a latching button so Step is either on or off. 
Still press n hold Step screen key to select the number of steps but also eliminate the 0 setting. From my original post I still like the idea of being able to select every fourth (eg) cell but it might require some button cocktail to perform.