polyend/TrackerBetaTesting #439: Jogwheel resolution/ increments

tfspsound, 2020-11-29:
Anyone tried going from fastest attack to slowest attack on the limiter? I had to take a break midway through. 
Any chance you could make the jog wheel move the play head about quicker? On the highest zoom (104, 96?)setting I seem to move 0.001s for every 3 clicks of the wheel. Surely it should move every click? At the other end of the spectrum i find myself spinning the jogwheel like a mad man and getting nowhere when scrolling through a sample even on lowest zoom (1) Not sure how long poor jog wheel can take it. 
The zoom function helps it just needs a tweak I think so it feels a bit more responsive, less spongey. 
Or a shift function to move 10’s or 100’s at once.

keith-macleod, 2020-11-29:
Switching display between samples/seconds/bars:beats would be bonus for those that are a bit OCD, such as myself.

mxactvtd, 2021-02-07:
my jogwheel is fairly unstable, not sure if its a software or hardware bug